glass-bead-labs/sensor-group #1: Ask for consulting on the university network, sensor network, experimenting side of things

davclark, 2015-02-09:
Should be assigned to @postfalk - who is working on this

postfalk, 2015-02-09:
From Aaron Culich:

[Domenico Caramagno](http://citris-uc.org/person/domenico-caramagno/) in CITRIS probably has the most direct experience with this in collaboration with faculty research in EECS who instrumented SDH (Sutardja Dai Hall) related to work in sensor networks and [Software-Defined Buildings](http://sdb.cs.berkeley.edu/sdb/), such as [SinBerBEST](http://crest.berkeley.edu/sinberbest.html) and LoCal.

Because CITRIS is part of EECS which runs its own wifi and wired networks the policies may be different than what campus more generally provides. For discussion with campus network folk, the best person to talk with on that front is [Isaac Orr in the IST-Networks](https://calnet.berkeley.edu/directory/details.pl?uid=960095) group.

At a minimum you should review:

[MSSND (Minimum Security Standards for Networked Devices)](https://security.berkeley.edu/mssnd)

And for devices that do not conform to those minimum standards (which is likely given the types of experimental devices you probably want to run), you can ask for an exception to be granted:

[Request for Exception: Berkeley Campus Minimum Security Standards](https://security.berkeley.edu/node/265)

Hope that helps!

postfalk, 2015-02-09:
@kevkoy We probably need to name/establish a security contact at the BIDS if we go down this route. Another option would be to ask the person (at the library) who is in charge of our network anyway to take this on which would be one IP for now.

davclark, 2015-02-09:
I'm already a campus security contact for D-Lab. I can be this.

davclark, 2015-02-09:
Also - we may want to rope the Siemens folks into this stage before we settle on protocols. That said, I'm guessing Bluetooth LE will be the fastest route, as I've started messing with this on the raspi, and one of our undergrads has also bluetooth experience.

postfalk, 2015-02-09:
Yes, just saw Aaron, he also suggested something other than Wifi. Bluetooth
is probably great. However, still attached to an Ethernet, Wifi subnet just
since it does not need any other hardware. On the other hand, we could just
do Arduinos with Bluetooth shields which might be cheaper.

Falk

closed 2015-02-12 by postfalk