glass-bead-labs/sensor-group #3: Basic supplies

davclark, 2015-02-09:
- [ ] ask whether we could use one of the little fish bowls as a hacker station (which can be still used for other things but will hold a modest amount of stuff - at home I get away with three boxes)
 - [ ] scope / get some basic supplies

postfalk, 2015-02-09:
*BASIC SUPPLIES*

A small collection of products:

* very basic
http://www.adafruit.com/product/136
https://www.sparkfun.com/products/13087

* somewhat advanced

* advanced

* other must haves
starter kit parts, https://www.sparkfun.com/products/10003
resistors, https://www.sparkfun.com/products/10969
breadboards, https://www.sparkfun.com/products/12002
wires, https://www.sparkfun.com/products/124

* sensors

postfalk, 2015-02-10:
We decided to create a shopping list not exceeding $1000. Kevin gave green light.

postfalk, 2015-02-10:
@davclark Schedule work meeting for finalizing list in BearBuy. Next week? After some more research.

davclark, 2015-02-10:
Generally, if my calendar is free, you are welcome to book it. I think this
should be doable in about an hour.

dani-lbnl, 2015-02-10:
Can we add some more stuff:
- soldering metal (white soldering, lead free?) and the solder
- batteries
- pliers and wire cutters (if not as part of any starter kit, usually not)
thx
Dani

Best Regards,
Dani
-------------------------------
Daniela Ushizima, PhD.
Head Deputy of the Visualization Group
Research Scientist, Lawrence Berkeley National Lab.
BIDS Data Scientist, University of California, Berkeley
1 Cyclotron Road MS 1650
Berkeley, CA 94720
lab: +1 510 486-4061
dushizima@lbl.gov

postfalk, 2015-02-10:
@dani-lbnl Does look sufficient for a first part http://www.adafruit.com/product/136, if not you could make wishes.

postfalk, 2015-02-12:
They are ordered. Issue will be closed when they arrive :+1:

davclark, 2015-04-01:
@davclark and @postfalk need to follow up with suppliers to see where still-unrecieved packages are.

closed 2015-04-29 by postfalk