glass-bead-labs/sensor-group #5: Address privacy concerns/develop policies

postfalk, 2015-02-10:
Since we are moving on this, I heard some concern recently. I think we should take following steps

- [ ] discuss among BIDS fellows
- [ ] develop policy
- [ ] design sign that makes people aware of sensors and what they do & collect
- [ ] make data streams public
- [ ] create repo where people can address concerns as GitHub issues

postfalk, 2015-02-12:
At our today's, fellow meeting we came to the conclusion that their is only very limited support for automatically collecting data on the use of the BIDS space in DOE 190. Fellows suggested all kinds of well-thought-out policies to deal with these issues. 

However, me (@postfalk) and @davclark do not have the capacity to work and implement such procedures and we doubt that they will make people happy. 

For this reason we decided to focus on environmental sensing for now.

davclark, 2015-02-13:
I'm going to propose a privacy / policy / ethics session this semester in the Tech4Measurement seminar I'll be organizing w/ the Social Science Matrix this year (we're having a planning lunch next Wednesday at 12:30 if you'd like to join). I'm going to call this essentially migrated off of our development path. Re-open if you'd like to continue the conversation here!

closed 2015-02-13 by davclark