glass-bead-labs/sensor-group #7: Get URAP students signed up for space-hackers

davclark, 2015-02-10:
### GitHub

Please leave a comment below with your username!

If you don't have a github account yet, please make one! And don't forget to "request a discount" for 5 free private repos (for your own projects) [here](http://education.github.com).

Once you do have a github id, please post to the above issue, and I'll add you to the team so we can collaborate on code. Don't worry if you don't know how to use github yet - this will be part of the onboarding process. If you're eager to learn more now, check the help button at the top of this page!

### Scheduling

In an ideal world, you all use google calendar to manage your time, and we'll use the "find a time" feature to schedule meetings. Let me know whether you do. If not, I will appoint one of the people who *doesn't* use google calendar to manage meetings via when2meet (or similar).​

### Lab Website

Also, feel free to make a pull request [here](https://github.com/glass-bead-labs/bead.glass/blob/gh-pages/members.md), which will update [the lab members page](http://bead.glass/members).

### Learning contracts

Once you've signed up here, we'll start implementing learning contracts for this semester.

jesscxu, 2015-02-10:
Hey this is Jessica! My username is: [jesscxu](https://github.com/jesscxu).

dianecw, 2015-02-10:
Hi this is Diane, username: dianecw

KishPatel1996, 2015-02-10:
Hey guys, I'm Kishan Patel.  Username:KishPatel1996

davclark, 2015-02-13:
Thanks everyone! I'll be getting learning contracts out over the weekend, and @sbenthall will be reaching out to organize our first in-person to get folks onto Slack for group communication, and start getting our hands dirty! Ideally, though, sign up ahead of time here:

https://glass-bead.slack.com

closed 2015-02-13 by davclark