glass-bead-labs/sensor-group #8: Things to learn for space sensing

postfalk, 2015-02-10:
OS

* Ubuntu/Debian
* OpenWrt
* smaller more embedded Debian versions, e.g. Raspbian

Microcontrollers

* Arduino
* Raspberry PI

Networking

* Wifi
* Bluetooth
* Ethernet

Coding

* Python
* some C (without any ++ or #), however the Wire.h and the Arduino derivatives provide some easy higher level functionality to deal with Microcontrollers

Protocols

* http
* I2C

Databases

* SQLite

Framework and Concepts

* ORM
* RESTful APIs
* Morepath (Python web framework, WSGI application)

Provisioning and Deployment

* Ansible 
* SSH
* ReverseProxy (Nginx)

davclark, 2015-03-25:
Looking at David Culler's site, and cross-correlating with my work on my own raspberry pi stuff, I found this in Lua:

http://eclipse.org/mihini/

I'm thinking lua might be a really good choice for us.

postfalk, 2015-03-25:
I am not sure whether we should add Lua. Makes it more complex and I am not
really fond of it. Works well on very small devices (and is part of
OpenWrt) but I am not sure whether our use case actually applies since the
Arduino UNO does not run an embedded Linux (Arduino Yun does) and we have a
full grown computer as gateway device. But of course we could tinker around
with it. But if you want to use a framework that already exists go for it.

Falk

closed 2015-05-08 by postfalk