glass-bead-labs/sensor-group #10: Requirement specification

postfalk, 2015-02-12:
Dear all,

I am looking forward to meet you all next week and I am very excited to get started on this project. @davclark and @postfalk ordered a small computer, Arduinos, blue tooth equipment, and sensors. I think we did a pretty good job and will be able to work hands-on as soon as the boxes arrive.

Here is a first pass on our thoughts on requirements and specs which implicitly guided or purchasing decisions. We will create a light weight sensor network in our BIDS space which can be easily deployed in many contexts.

**Part 1:**

Deploy gateway computer as server for sensors, sensor API, and basic web application

**Part 2:**

Create an universal Arduino sketch that can communicate with a server via BlueTooth. Create plug-able code that can communicate with a particular sensor. Keep track of the deployment of the sensors in time and space. Create room coordinates and meta data framwork.

**Part 3:**

Create application that frequently stores sensor reading into a database.

**Part 4:** 

Create web application, RESTful API to serve sensor data to the Internet. Create some basic visualizations of the data streams to present. 

**Part 5:**

Scale sensor network up and try to interpret sensor streams in the sense of activities. Explore machine learning algorithms for this task.

**Part 6:**

Create web application that supports supervised classification by users.

davclark, 2015-02-17:
From the meeting today: can we implement something like a "Do Not Track" option in some cases?

postfalk, 2015-02-18:
I think at this point we should stick with the plan of exploring the full stack to identify the problems along the data travel which are directly connected to the data model/provenance question. Let's see with what practical ideas our students come up with (have surely some backup) and give work assignments for students where they are interested/knowledgeable.

postfalk, 2015-03-04:
Duplicates #31

closed 2015-03-04 by postfalk