glass-bead-labs/sensor-group #13: Investigate using mobile phone as sensor or bluetooth hub for our dedicated sensors

davclark, 2015-02-18:


KishPatel1996, 2015-02-23:
#### Using phones as sensors:

###### Android
There is a pretty interesting article on [Instructables] (http://www.instructables.com/id/Use-your-android-phone-sensors-on-the-arduino-/?ALLSTEPS)  about connecting an Arduino to an Android handheld to gather data and access its sensors.

###### iPhone
Apple has more restriction on what we can do with its sensors.  However, we can access data already logged in Apple's Health app using [HealthKit](https://developer.apple.com/healthkit/).  If we develop an app, we would be able to access this information, but it is limited in scope to what the phone user has allowed the Health app to log.  Also, we would need a develop account ($99/year) to run the app on an iPhone.  Without the developer account, we are limited to an iPhone simulation on a laptop.  

I think using Android phones as sensors would be the most feasible option.

Still looking for some good readings on using phones as a bluetooth hub.

postfalk, 2015-02-23:
Thank you, did the same thing and I think the Android route sounds most
reasonable to get started. Who in the group has an Android?

Falk

dianecw, 2015-02-24:
I have an Android phone as well as an Android tablet- I can bring both to the work session tomorrow.

postfalk, 2015-02-24:
Great. We will probably not get to that tomorrow. However, if our shipment
is there you could get started on your own.

I think tomorrow we start to work with APIs.
Next week basic arduino.
After that we will create our first basic implementations.

Just eye-balled this should talk about timeline at next meeting.

See you tomorrow.

Falk