glass-bead-labs/sensor-group #14: Collect list of on-campus maker resources

davclark, 2015-02-18:
For example, Invention lab, Marvell Nanolab, Social Apps Lab (get links!). This is a task for everyone.

dianecw, 2015-02-18:
[CITRIS Invention Lab](http://invent.citris-uc.org/) - Rapid prototyping space with 3D-printers, laser cutters, etc.
[Marvell Nanolab](http://nanolab.berkeley.edu/) - Micro- and nano- fabrication tools for creating sensors.

postfalk, 2015-02-18:
@davclark Do you think we could get an introduction and access to CITRIS?

davclark, 2015-02-20:
This is apparently free (and @arjunmehta94 in the taichi group is already a member!): http://supernode.berkeley.edu

dianecw, 2015-02-23:
I just wanted to leave a note that I just won access to the CITRIS invention lab through a hackathon, and will have access this semester. It also sounds like they are pretty lax about bringing other people into the lab, so I can swipe one or two people in with me anytime!

I'll be doing safety training on Wednesday, so I can give more feedback after then.

davclark, 2015-02-23:
Congrats Diane! And I look forward to checking out the invention lab...