glass-bead-labs/sensor-group #17: Write adapter for GitHub API to ASBase

davclark, 2015-02-24:
- [ ] Pull data from github API for `https://github.com/glass-bead-labs` repositories using requests
 - [ ] Post data to ASBase

Can you see this issue? Can you see the commit that solves the issue (and each sub-component)? Can you see whether the issue was automatically closed or whether it was manually closed?

davclark, 2015-02-24:
In collaboration with @KishPatel1996.

dianecw, 2015-02-25:
Made a basic issue tracker that pulls issues from this repository and posts basic information about the issue to ASbase. Doesn't cover information about solving issues/closed issues.

smnmyr, 2015-02-25:
Sounds great! Is that information posted to ASbase already? What's its basic structure?

dianecw, 2015-02-25:
Yep, we posted the current issues in this sensor-group repository to the database. Every time we run our python script, it posts issue creator info (github url, creator id, creator github username), issue url/id, and repository info (github url, id, name) to the database. This is in the data format specified on the ASbase API.

Right now, we're working on functionality to only post new data if there's an update to the issue.

KishPatel1996, 2015-02-25:
There seems to be an issue with the way the Activity Stream is displaying the target information

Here is an example: 
Actor davclark does/did/had issue with/to 57076355 at 19:20:08 (2015-02-09)

Instead of taking the name of the target, it seems to be displaying the id even though the fields have been entered correctly.  Is this the way the stream was designed, or have we missed something?

smnmyr, 2015-02-25:
That behavior is intended. Indeed, the 57076355 thing is the object that davclark worked on, not the target (and the object doesn't have a displayName). Since the target is not used in lots of Activity Streams, the platform is not considering it in the digests on the front page - you are able to see the entire activity in the /query resource though, here: http://russet.ischool.berkeley.edu:8080/query/

One more comment about your activity itself: Is "issue" the verb you wanna go with? I'd say that you could be more expressive, depending on what you wanted to say:
* Did davclak raise an issue - then the verb could be "post", "raise", etc.
* Did he resolve an issue - the verb could be "post", "resolve", etc.
* etc.

Have a look at http://activitystrea.ms/registry/verbs/ to find if there is already something that could be used.

cool that the plumbing is working ;-)

smnmyr, 2015-02-25:
I updated the server to display more information about the activities (incl. the target) on the front-page digest.

KishPatel1996, 2015-02-26:
Ah, understood.  Right now, "issue" is a placeholder since we haven't added some functionality yet (checking for updates, seeing if any issues were closed, etc).  @dianecw and I will be meeting up tomorrow from 2 to 4 to complete this.  We'll add more descriptive words at that point.

Also, thanks for the activitystrea.ms link.  Really helpful!

davclark, 2015-02-26:
And along those lines, don't forget these recommendations from @smnmyr:

When you create activities to push to ASbase, adhere to a process along these guidelines (more details here: http://dret.net/lectures/iot-spring15/rest from slide 34)
 1. Try to use verbs and objects that are already defined in the activity schema, here http://activitystrea.ms/registry/verbs/
 2. If that doesn't feel fully right (but approximately), try to use one of the defined verbs, but extend it. For our project, we've added a "subVerb" property that extends (but not hijacks!) the base verb
 3. If it's just not possible to reuse an already defined verb without hijacking, create a new one (and document it!)

dianecw, 2015-02-27:
Quick Update: We got it working! The database updates when you run the script if any issues have been created, updated, or closed since the last post. We'll add continuous stream functionality later. Kishan is adding some verbs and functionality, but the basics are all there.

Note: I'm using a temporary personal authentication token, which needs to be reset after a certain amount of time. For the future, we should look into OAuth to authenticate our application.

KishPatel1996, 2015-02-27:
We have completed the git issue tracker program!  It updates the ASBase depending on whether an issue has been created (create), updated (update), and closed (close).  

The program currently runs every 10 seconds.  Is there a more efficient way to keep the program running other than a while loop?

```python
while True:
    updateASBase() #this gets the latest changes and posts them on ASBase
    time.sleep(10)
```

smnmyr, 2015-02-27:
Nice work! Since github doesn't have an event-based API, this is the way to do it ;-)

KishPatel1996, 2015-02-27:
Does the GitHub API support getting the user who last commented on an issue?

dianecw, 2015-02-27:
Looks like they do [here](https://developer.github.com/v3/issues/comments/), but it might take some time to figure out how it works.

KishPatel1996, 2015-02-27:
@dianecw   It seems like the code is now reporting "Bad credentials," most likely from the auth parameter of requests.get.

The code runs without the authentication, but it eventually gets this ...

'https://developer.github.com/v3/#rate-limiting', 'message': "API rate limit exceeded for 136.152.125.124. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.

Can you check what's going on with the authentication?  Pull from master again, as I have hashed out the requests with authentication and written requests without them.

dianecw, 2015-02-27:
Yeah, I added a note above. I got around that by generating a temporary authentication key from my account (you can do it too from your account settings -> authentication). I need to regenerate it every session. If we want a more permanent solution, we'll have to learn how to use OAuth, which I don't have experience with but am willing to learn if needed.

davclark, 2015-02-27:
I can give you some pointers on using oauth, that side of things can be hairy. Let me know if you want assistance!

postfalk, 2015-02-27:
Good work. There is a (at least one) Python library supporting oauth
(oauthlib) as well as an integration of this library into requests
requests/requests-oauthlib · GitHub
<https://github.com/requests/requests-oauthlib>

Falk

KishPatel1996, 2015-02-27:
I believe we need to "register a new OAuth Application."  Here is the link

https://github.com/settings/applications/new

postfalk, 2015-02-27:
That might or might not be correct. I think what you are looking at is a
way to use Github's authorization system as an authorization layer for a
third party app. You surely know that principle from web pages asking you
to log-in with facbook, google+, or github. This the core use case for
OAUTH.

What you want to do is to just login to github. More along those lines ...

https://developer.github.com/v3/#authentication

You can use just user/password authorization
https://gist.github.com/caspyin/2288960. In practice you should use the
user/passwd authorization only to request a token and use the token for
further interaction.

Saying this, I am not sure how we should go about this. Could you increase
the request on the public BIDS repo issues even if you authorized to
another account. I am not sure whether we should share the BIDS credentials
for this experiments.

Falk

davclark, 2015-02-28:
The standard approach I've seen other apps use is OAuth tokens (not username / password). That said, I'm going to suggest that while we are developing an "app" - it's not going to ask users to generate keys. I think probably the right approach is to use personal access tokens (accessible from the same root page):

https://github.com/settings/applications

"Personal access tokens function like ordinary OAuth access tokens. They can be used instead of a password for Git over HTTPS, or can be used to authenticate to the API over Basic Authentication."

postfalk, 2015-03-02:
That is correct. However, in some instances tokens expire, in this case you
might need to request a new token first. However, for github.com it looks
like you can create this token with the online form and it does not expire.
So Dav's way should work.

Falk

KishPatel1996, 2015-03-02:
ok, i generated a key and added it to the file.  Should work

davclark, 2015-03-02:
Generally, it's good practice to read keys from an external file, and not commit them to the repo. That said, as long as you don't mind the world having the permissions associated with that key, I don't think it's a problem, but it would be good to refactor the key into a separate file just to get into the practice.

KishPatel1996, 2015-03-02:
It seems that github revokes the Key if I commit it to a public repo, which most likely happened in the first case.

For reading from an external file, how would I set that up with the server in mind?

postfalk, 2015-03-02:
Your external key would be just a file in your project but you would
.gitignore it. Once it is there, you can get the key by

f = file('filename', r)
key = f.read().strip() # I put strip here to make sure that new line or
spaces do not interfere with the key's working.
f.close()

davclark, 2015-03-02:
Or, with more modern python:

```
with open('filename') as f:
    key = f.readline().strip()

# f is implicitly closed by context manager because of `with` statement
```

D

KishPatel1996, 2015-03-02:
Moved the authentication key to an external file.  

The file will need to be named  ```GitHubAuthKey.txt ``` and be in the  ``` github-api/ ``` directory.

Other than that, I think we are done.

KishPatel1996, 2015-03-04:
Should we be looking towards other APIs?  I know Slack has an extremely friendly API and we can probably implement it in a couple of hours top.

postfalk, 2015-03-04:
Sounds great. We could also think what kind of visualization would be
useful for the ASBase. We could start with something like circles for every
event and then assign properties specifying this event. We did not get to
d3.json yesterday but we could discuss this briefly next week (or on
Friday).

KishPatel1996, 2015-03-04:
OK, I'll work on that tomorrow (Thursday).  Maybe i can also create a Slackbot that posts updates to issues and such onto the #space-hackers group.

davclark, 2015-04-01:
This has been migrated to the [data-connector issue tracker](/glass-bead-labs/data-connector/issues/3).

closed 2015-04-01 by davclark