glass-bead-labs/sensor-group #20: Agenda for Feb 25, 2015 Meeting

davclark, 2015-02-25:
1. Report out on work-in-progress
    a. pulling data from GitHub API (general HTTP+JSON interfaces)
    a. work towards dynamic visualizations
    a. state of purchasing / sensors
 1. Plans for next week
    a. Whiteboard: overview of what we'll do with sensors (provide reading)
    a. Scheduling
 1. Make sure everyone is on Slack, GitHub, etc.

postfalk, 2015-02-25:
>    2. Plans for next week a. Whiteboard: overview of what we'll do with
>    3. Make sure everyone is on Slack, GitHub, etc.

A little bit chaotic but the important terminology is there.

https://en.wikipedia.org/wiki/Sensor

For the application level it means to create a data model that can deal
with crazy states of  sensors. Malfunction seems to be the ideal event for
detection.

We could also think a little bit about state vs. change (event, activity).

Looking forward to this,
Falk

davclark, 2015-02-25:
From @smnmyr:

I wanted to get back to you about the topic of activity modeling: When you guys create activities to push to ASbase, I think it'd be a good idea to make everyone adhere to a process along these guidelines (more details here: http://dret.net/lectures/iot-spring15/rest from slide 34)
 1. Try to use verbs and objects that are already defined in the activity schema, here http://activitystrea.ms/registry/verbs/
 2. If that doesn't feel fully right (but approximately), try to use one of the defined verbs, but extend it. For our project, we've added a "subVerb" property that extends (but not hijacks!) the base verb
 3. If it's just not possible to reuse an already defined verb without hijacking, create a new one (and document it!)

davclark, 2015-02-25:
Falk will run PostgreSQL workshop during the Wed 10am time slot (around March 28th and April 4th)

Folks focusing on web viz can float to other meetings, for example, the Monday 3pm meeting.

postfalk, 2015-02-25:
Should we close this now?

closed 2015-02-25 by davclark