glass-bead-labs/sensor-group #21: Agenda Tuesday 3/3

postfalk, 2015-02-26:
* Xmas - unpacking Arduinos and sensors
* Get LED 13 blinking (aka Arduino basics)
* Use other pins, introduction into C, Wire.h

postfalk, 2015-03-09:
@davclark that is the agenda for tomorrow. Anything else we should talk about it, yes we are starting at 3pm.

davclark, 2015-03-09:
I was pinging you because it wasn't on the calendar!

postfalk, 2015-03-09:
Oh sorry, I did not update the calendar. I should copy all these things to
the glass-beads calendar.
Did you read about my success with the blue-tooth. I think the cheap
modules still need pairing but it can be easily done from Python.
I hope to get a chance to implement this tomorrow.

That is where I am at (requires still manual setup of the bluetooth
connection).
https://gist.github.com/postfalk/604d667da192a86d8534

In this code there is actually no bluetooth handling at all (except the
access to the serial socket). It just uses the serial that is layered on
top of it.

Falk

postfalk, 2015-03-10:
https://learn.adafruit.com/photocells

postfalk, 2015-03-11:
Here is the math

http://home.roboticlab.eu/en/examples/sensor/photoresistor

postfalk, 2015-03-11:
https://en.wikipedia.org/wiki/I%C2%B2C

closed 2015-03-11 by postfalk