glass-bead-labs/sensor-group #22: ASBase Queries All Failing

KishPatel1996, 2015-02-26:
We are having trouble querying information from the ASBase.  

Can someone check if this is an issue with the database, or are we using it improperly? 
@smnmyr  @davclark 

 **Here are some examples that we have tried, all of which are returning "request failed."** 
http://russet.ischool.berkeley.edu:8080/query?startTime=2015-02-24
http://russet.ischool.berkeley.edu:8080/query?startTime=2015-02-22&endTime=2015-02-24

We also tried advanced queries using the MongoDB language in JSON format.

smnmyr, 2015-02-26:
Fixed. Thanks!

dianecw, 2015-02-26:
Looks like all the queries are now returning
"{"totalItems":0,"@type":"http://activitystrea.ms/2.0/Collection","items":[]}".

smnmyr, 2015-02-26:
Yes, the query http://russet.ischool.berkeley.edu:8080/query?startTime=2015-02-24 returns 0 items. I cannot find a mistake, however. Did you put activities into the DB with a startTime > 2015-02-24?

I can find your uploads using the query `POST http://russet.ischool.berkeley.edu:8080/query`
and contents `Content: { "verb" : { "$in" : [ "issue" ] } }`

dianecw, 2015-02-27:
Great, I think I got it working. Thanks for your quick responses!

closed 2015-02-27 by dianecw