glass-bead-labs/sensor-group #27: Today's introduction in D3.js

postfalk, 2015-03-03:
Since we decided to do this yesterday it will be somewhat a ad-hoc introduction. I will mostly cover the DOM manipulation capabilities of D3.js (and exemplify them with some graphic examples). 

[Three little circles] (http://bost.ocks.org/mike/circles/) is always a good start. 

Javascript is a somewhat crazy language mainly because it has an enormous flexibility. You can easily get started in a few hours but it takes a lot of time to master it. Always a good video to see what you get self into http://googlecode.blogspot.com/2009/03/doug-crockford-javascript-good-parts.html. It used to be a five part thing but now you have to pay for the full version, bummer.

Here are a few things, we will probably not touch today but they are good to know (or at least heard of) when you start to deal with JavaScript ...

* client-side vs. server-side code execution
* static web pages vs. web applications
* DOM manipulation
* JS frameworks, such as JQuery, D3, and many more
* scope (aka pain in the butt)
* asynchronous code execution/loading
* events
* software patterns, e.g. singleton
* prototypical inheritance

dani-lbnl, 2015-03-03:
Where will it be? may i join.

postfalk, 2015-03-03:
Sure it will be at 4pm (we meet at BIDS at 3:30 but will really start at 4pm). It will be really basic though since we did not decide before yesterday to do it.

Falk

postfalk, 2015-03-04:
**More d3.js**

It was fun to do some d3.js today. I would like to point out a few more resources.

On Friday 3/13, Kai Chang of [Stamen Design] (http://stamen.com/) will present his work at our weekly Data Science Lecture. In the second half of the two hour meeting, he will life code a new visualization from scratch using [Ecoengine](https://ecoengine.berkeley.edu) Data (an API I have been developing). The presented techniques and workflows will be very useful for our sensor project.

postfalk, 2015-03-04:
There is also a meetup group providing introductions and presentations on advanced topics:

http://www.meetup.com/Bay-Area-d3-User-Group/

postfalk, 2015-03-04:
Here is the cleaned up code from yesterday's session.

http://bl.ocks.org/postfalk/f0d3ceddd49f0c6fa029

postfalk, 2015-03-04:
Moved to Wiki

closed 2015-03-04 by postfalk