glass-bead-labs/sensor-group #29: Doc strings are usually placed underneath the first line of a function/class declaration

postfalk, 2015-03-04:
```python

def some_function():
     """This function does nothing."""
     pass
```

From this documentation and even unit tests can be generated automatically. You could stuff your entire documentation text there (which is only feasible for smaller projects.) See https://www.python.org/dev/peps/pep-0257/

davclark, 2015-03-06:
What is the issue-tracker label for? Maybe we should have a wiki page about our labelling conventions?

postfalk, 2015-03-07:
issue-tracker is for the actual github-issue-tracker, people are working
on. We are getting in some self-referential problems here. For this reason
we should move actual software in a different repo.

Falk

KishPatel1996, 2015-03-12:
This issue was moved to glass-bead-labs/data-connector#2

closed 2015-03-12 by KishPatel1996