glass-bead-labs/sensor-group #30: Agenda Tuesday 3/17

postfalk, 2015-03-04:
Create first version of a dashboard

postfalk, 2015-03-16:
Hi, 

... a quick update on the Agenda. I did not succeded in creating a very good project structure for the data collector yet. I am still experimenting with a setup over serial that allows for painless and automatic reconnecting, a timed measurement, and error reporting.

For tomorrow, we will work on the dashboard. I think we will mostly connect our d3 example to some real data from ASBase. That will be a lot of fun.

For anyone who would like to dash ahead, you could look at 

https://github.com/mbostock/d3/wiki/Requests

Don't get scared just scroll down to d3.json and remember callback functions that is all what is needed.

Something like request, reformat data, call update should get your data on the screen

Falk

davclark, 2015-03-17:
Am I running this, or are you, @postfalk?

postfalk, 2015-03-17:
I could run it or you could. I will be there. I could also just help with
the d3.js part and work on my serial part. My comment was more about what I
didn't do over the week and that we will be ok anyways.

What is your preference? I will be happy to lean back.

Falk

jesscxu, 2015-03-17:
http://russet.ischool.berkeley.edu:8080/query/?startTime=2015-02-24

postfalk, 2015-03-17:
https://github.com/mbostock/d3/wiki/Requests

postfalk, 2015-03-18:
http://bl.ocks.org/postfalk/f0d3ceddd49f0c6fa029

postfalk, 2015-03-18:
https://decyborg.wordpress.com/2013/09/08/bluetooth-serial-communication-with-arduino-jy-mcu-bluetooth-and-macbook-pro/

postfalk, 2015-03-18:
https://gist.github.com/postfalk/604d667da192a86d8534

postfalk, 2015-03-18:
http://www.jeromecukier.net/blog/2011/08/11/d3-scales-and-color/

closed 2015-03-18 by postfalk