glass-bead-labs/sensor-group #31: Agenda Tuesday 3/24

postfalk, 2015-03-04:


postfalk, 2015-03-06:
Quick check-in: Is everyone ok with the fact that this very important session is during spring break or should we move it?

dianecw, 2015-03-06:
I'll be home for spring break (3/21-3/29), so I won't be able to make it. Could this meeting be moved earlier? Even though I'll be home, I'll be spending most of my time working so it'd be great to have things to work on over break.

postfalk, 2015-03-06:
What about Friday 3/20 in parallel to the Tai Chi group (2pm-5pm). Cannot
confirm yet (depends on the Postgres intro).

postfalk, 2015-03-06:
Should we move this to 3/20 2pm to 4:45pm. I could confirm this time if everyone would be in board.

dianecw, 2015-03-06:
This is fine with me, thank you!

postfalk, 2015-03-06:
Let's see whether it works for @KishPatel1996 and @jesscxu as well. And confirm next week.

jesscxu, 2015-03-06:
That is fine with me too!

KishPatel1996, 2015-03-06:
I have a flight from SFO at 9:30 on the 20th.  2 to 4:45 should be fine.

postfalk, 2015-03-20:
virtualenv -p /usr/bin/python3 env

postfalk, 2015-03-20:
https://ecoengine.berkeley.edu/

postfalk, 2015-03-24:
@jesscu Are you still planning on coming in today? I just would like to now and decide where to work today. There is a lot of stuff going on at BIDS.

closed 2015-03-24 by postfalk

jesscxu, 2015-03-24:
I figured out the problem I had earlier, so I won't be coming in today!

postfalk, 2015-03-24:
Ok, thanks. Enjoy spring break.